n-serrette/esphome_sd_card #2: Add support for read file and list directory

andreitoma-github, 2024-09-22:
Hello, the external component works perfect for write ,append and delete file, create and delete folder

Would be great if you can add support for read file and list directory with a configurable depth

Thanks!

n-serrette, 2024-09-23:
Hi,

Thanks for your interest, I have this two functionality planned but I lack of time a the moment. I will try to work on this when i can.

Could you describe your use case?

n-serrette, 2024-09-23:
I forgot but the component already expose a function list_directory returning
```cpp
std::vector<std::string> list_directory(const char *path, uint8_t depth)
```

you can use it like that:
```yaml
- lambda: |
    for (auto const & file : id(esp_camera_sd_card)->list_directory("/", 1))
      ESP_LOGD("   ", "File: %s\n", file.c_str());
```

do you need it to return more than the files path?

n-serrette, 2025-01-01:
I have added support, to read file

```cpp
std::vector<uint8_t> read_file(char const *path);
std::vector<uint8_t> read_file(std::string const &path);
```

Return the whole file as a vector, trying to read large file will saturate the esp memory

* **path**: file path

Example

```yaml
- lambda: return id(esp_camera_sd_card)->read_file("/file");
```

closed 2025-01-01 by n-serrette