n-serrette/esphome_sd_card #6: Usage

holefrog, 2024-12-19:
Hi,

Thank you for your component. I am trying to use it with my ESP32-CAM board. 

I formatted the 8G micro-SD card using Fat32. I can get the picture of the camera, however the values of 'SD card used space', 'SD card free space', and 'SD card type' are 'unknown'. I am not sure what the issue is, might be a defective board or not.

Below is my yaml:
```
......

external_components:
  - source: github://n-serrette/esphome_sd_card

sd_mmc_card:
  id: esp_camera_sd_card
  mode_1bit: false
  clk_pin: GPIO14
  cmd_pin: GPIO15
  data0_pin: GPIO2
  data1_pin: GPIO4
  data2_pin: GPIO12
  data3_pin: GPIO13

esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz

  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32

sensor:
  - platform: wifi_signal
    name: '$hostname WiFi Signal'

  - platform: sd_mmc_card
    type: used_space
    name: "SD card used space"

  - platform: sd_mmc_card
    type: total_space
    name: "SD card total space"

text_sensor:
  - platform: sd_mmc_card
    sd_card_type:
      name: "SD card type"

```

The logs show that "Component sd_mmc_card is marked FAILED"
```
[15:19:39][C][sd_mmc_card:018]: SD MMC Component
[15:19:39][C][sd_mmc_card:019]: Mode 1 bit: 0
[15:19:39][C][sd_mmc_card:020]:   CLK Pin: GPIO14
[15:19:39][C][sd_mmc_card:021]:   CMD Pin: GPIO15
[15:19:39][C][sd_mmc_card:022]:   DATA0 Pin: GPIO2
[15:19:39][C][sd_mmc_card:024]:   DATA1 Pin: GPIO4
[15:19:39][C][sd_mmc_card:025]:   DATA2 Pin: GPIO12
[15:19:39][C][sd_mmc_card:026]:   DATA3 Pin: GPIO13
[15:19:39][C][sd_mmc_card:029]:   Used space 'SD card used space'
[15:19:39][C][sd_mmc_card:029]:     State Class: 'measurement'
[15:19:39][C][sd_mmc_card:029]:     Unit of Measurement: 'B'
[15:19:39][C][sd_mmc_card:029]:     Accuracy Decimals: 2
[15:19:39][C][sd_mmc_card:029]:     Icon: 'mdi:memory'
[15:19:39][C][sd_mmc_card:030]:   Total space 'SD card total space'
[15:19:39][C][sd_mmc_card:030]:     State Class: 'measurement'
[15:19:39][C][sd_mmc_card:030]:     Unit of Measurement: 'B'
[15:19:39][C][sd_mmc_card:030]:     Accuracy Decimals: 2
[15:19:39][C][sd_mmc_card:030]:     Icon: 'mdi:memory'
[15:19:39][C][sd_mmc_card:037]:   SD Card Type 'SD card type'
[15:19:39][E][component:082]:   Component sd_mmc_card is marked FAILED
```

n-serrette, 2024-12-20:
Hi,

Thanks for your interest in this project.
After testing their is an issue with the component in 4 lane mode, I will look into it.
In the meantime, you can try the 1 bit mode wish seems to work as intended.

n-serrette, 2024-12-20:
After some investigation it seems the issue come from the latest arduino framework version (>3.0.0), using  the version 2.0.17 solve the issue.

```yaml
esp32:
  board: esp32dev
  framework:
    type: arduino
    version: 2.0.17
```

I will try to see what need to be done to migrate to the latest version of the arduino framework

holefrog, 2024-12-20:
> After some investigation it seems the issue come from the latest arduino framework version (>3.0.0), using the version 2.0.17 solve the issue.
>
> ```yaml
> esp32:
>   board: esp32dev
>   framework: [...]

Hi, 
Thank you for your quick response.
It works after changing frame work to 2.0.17 manually. Now, I can get SD card information. I am going to try other functions.

Merry Holidays,

n-serrette, 2025-01-21:
I have passed more time on the issue and I was totally mistaken with my interpretation.

The problem does not come from the 3.0.0 version of the arduino framework, from my understanding it is not possible to use it with esphome.

At the moment the arduino framework version used by esphome are:
* **recommended**: 2.0.5
* **latest**: 2.0.9

From that point I have found the pull request resolving the issue [arduino-esp32/#7646](https://github.com/espressif/arduino-esp32/pull/7646) in the version 2.0.7

I will update the documentation to reflect this findings

closed 2025-01-21 by n-serrette