n-serrette/esphome_sd_card #7: Read mp3 file from the SD card

nbucic, 2024-12-27:
Hi @n-serrette ,

Can I create esphome media_player that would pool files locally from the SD card, using your library for reading sd card files?

n-serrette, 2024-12-27:
Hi,

Thanks for your interest, at the moment it is not possible, the component does not support reading from the sd card yet.
I am working on the reading aspect, but progress is slow because I have limited time at the moment.

You have a great use case, I take it into account while developing the reading and try to implement it.

I have never use the audio aspect of esphome, could you tell me what hardware you plan to utilize?

nbucic, 2024-12-28:
Hi @n-serrette,

Thank you for showing the interest in my use-case.

I'm currently using ESP32 and ESP32-S3 (with PSRAM), together with MAX98357 DAC. 
I have stored some mp3 files on my server and the stream from the internet is working without any issue. 

In the long run, I would like to have stored these files on the SD card.

Before switching to the ESPHOME scripting, I wrote Arudino code that worked following this tutorial: https://dronebotworkshop.com/esp32-i2s/

I very unfamiliar with the C/C++ code (know to read and understand it, hard to write anything new), so I can't help you on your quest. But this is a very good and promising library.

n-serrette, 2025-01-01:
Hi,

Thanks for the detail, I will try to procure some hardware to develop this but I cannot make promise en the timing.

In the mean time I have added a simple read_file function.

```cpp
std::vector<uint8_t> read_file(char const *path);
std::vector<uint8_t> read_file(std::string const &path);
```

Return the whole file as a vector, trying to read large file will saturate the esp memory

* **path**: file path

You can try playing a file with the [Speaker Component](https://esphome.io/components/speaker)

```yaml
- speaker.play:
      id: my_speaker
      data: return id(esp_camera_sd_card)->read_file("/file");
```

nbucic, 2025-01-02:
Hey, thank you very much for the addition. Will try to find some time to setup sdcard together with esp32-s3 and will let you know the outcome.

TarheelGrad1998, 2025-02-11:
I've been searching for similar functionality, but this isn't working for me:

Getting:
          Expected integer, but cannot parse return id(sd_card)->read_file("/file.mp3"); as an integer.

n-serrette, 2025-02-13:
Hi,

Sorry I did not try my example, it's missing the ````!lambda```` before the return.
The correct version should be:
````yaml
- speaker.play:
    id: my_speaker
     data: !lambda return id(sd_card)->read_file("/file.mp3");
````

youkorr, 2025-03-03:
how did you manage to make the mp3 playback work because I already have a media player pe on esphome it refuses any attempt - speaker.play:
id: my_speaker
data: !lambda return id(sd_card)->read_file("/file.mp3");

TarheelGrad1998, 2025-03-03:
Me, I could never get this to work, unfortunately.  I'm using the Waveshare ESP32-S3-LCD-1.85, and I hit two issues:

- The device has NC for the D1 and D2 pins.  The component seems to require them?
- This component doesn't seem to support the EXIO pin, at least, this format would not compile:
`      pca9554: pca9554a_device
      number: 2`

youkorr, 2025-03-03:
on mine esp32s3box3 at first it was impossible for me to activate the sd reader by checking I was missing the SDCARD_PWR_CTRL gpio 43 so I found another solution now we can see the sd card, but I can't get the audio file on the sd card to work

youkorr, 2025-03-03:
[image]

youkorr, 2025-03-03:
`sd_mmc_card:
  id: box3_sd
  clk_pin: GPIO11
  cmd_pin: GPIO14
  data0_pin: GPIO9
  data1_pin: GPIO13
  data2_pin: GPIO42
  data3_pin: GPIO12
  mode_1bit: false
  power_ctrl_pin: GPIO43`

n-serrette, 2025-03-03:
> * The device has NC for the D1 and D2 pins.  The component seems to require them?

D1, D2, D3 are not required to use the component in 1bit mode you have to set `mode_1bit: true`

> * This component doesn't seem to support the EXIO pin, at least, this format would not compile:
>       `      pca9554: pca9554a_device number: 2`

As far as I know sdmmc pin mapping is link to the esp board version and can't use io expander

youkorr, 2025-03-03:
I am working on a project but there is a lot to do here is an example  

`storage:
  - platform: sdmmc
    id: my_storage
    sdmmc_id: sd_card
    files:
      - source: /sounds/test.mp3
        id: box3_sd`

n-serrette, 2025-03-04:
Could you open a new issue and give more detail on your project so this issue stay centered on playing audio.

thanks

youkorr, 2025-03-05:
here are the logs no sound plays
`[18:12:06][I][ALARM_AUDIO:284]: File read, size: 87093 bytes
[18:12:06][I][SPEAKER:289]: Attempting to play file, size: 87093
[18:12:06][D][ring_buffer:034][speaker_task]: Created ring buffer with size 16000
[18:12:06][D][i2s_audio.speaker:111]: Starting Speaker
[18:12:06][D][i2s_audio.speaker:129]: Stopped Speaker
[18:12:06][W][i2s_audio.speaker:143]: Error writing to I2S: ESP_ERR_INVALID_STATE`

youkorr, 2025-03-06:
implement them to work properly with audio and image files.
These implementations:

Check if the path is valid
Open the file in binary mode
Determine the file size
Allocate a vector of the correct size
Read the entire contents of the file into the vector
Handle possible errors

This approach works for any type of file, including audio and image files, because it simply reads the raw binary data. Format-specific processing (audio decoding, image display, etc.) will have to be done elsewhere in your code after you retrieve this raw data.
This implementation is compatible with your ESP Home architecture and should work on both ESP32 and ESP8266 platforms.

`std::vector<uint8_t> SdMmc::read_file(char const *path) {
  if (path == nullptr) {
    ESP_LOGE("sd_mmc", "Cannot read file: path is null");
    return {};
  }

  // Vérifier si le fichier existe
  if (!this->is_directory(path)) {
    FILE *file = fopen(path, "rb");
    if (file == nullptr) {
      ESP_LOGE("sd_mmc", "Failed to open file %s for reading", path);
      return {};
    }

    // Obtenir la taille du fichier
    fseek(file, 0, SEEK_END);
    size_t file_size = ftell(file);
    fseek(file, 0, SEEK_SET);

    // Créer un buffer de la taille du fichier
    std::vector<uint8_t> buffer(file_size);

    // Lire le contenu du fichier
    size_t bytes_read = fread(buffer.data(), 1, file_size, file);
    fclose(file);

    if (bytes_read != file_size) {
      ESP_LOGE("sd_mmc", "Failed to read file %s completely, expected %zu bytes but got %zu", 
               path, file_size, bytes_read);
      buffer.resize(bytes_read);  // Ajuster la taille du buffer si nécessaire
    }

    ESP_LOGD("sd_mmc", "Successfully read %zu bytes from file %s", bytes_read, path);
    return buffer;
  } else {
    ESP_LOGE("sd_mmc", "Cannot read %s: it's a directory", path);
    return {};
  }
}

std::vector<uint8_t> SdMmc::read_file(std::string const &path) {
  return this->read_file(path.c_str());
}`

youkorr, 2025-03-06:
but i know if it works