n-serrette/esphome_sd_card #10: Logging Sensor Data to SD Card

TCB13, 2025-02-13:
Hello,

Can you provide a quick example on how one can log data from a sensor to the SD Card?

Thank you.

n-serrette, 2025-02-13:
Hi,

Thanks for your interest, I don't have this sensor but it should look something like this:

```yaml
sd_mmc_card:
  id: sd_card
  mode_1bit: false
  clk_pin: GPIO14
  cmd_pin: GPIO15
  data0_pin: GPIO2
  data1_pin: GPIO4
  data2_pin: GPIO12
  data3_pin: GPIO13

i2c:

bme680_bsec:

sensor:
  - platform: bme680_bsec
    temperature:
      name: "BME680 Temperature"
      on_value:
        then:
          - sd_mmc_card.append_file:
              path: "/temperature.log" 
              data: !lambda |
                std::string str = esphome::to_string(x) + "°C\n";
                return std::vector<uint8_t>(str.begin(), str.end());
```

Hope this can help you.

TCB13, 2025-02-13:
Thank you very much!

closed 2025-02-28 by n-serrette