n-serrette/esphome_sd_card #11: Using esphome_sd_card for device  LILYGO T3_V1.6.1

paradisgaston, 2025-02-15:
Hi, I wish to read/write sensors data to SD Card. 
My device is LILYGO T3_V1.6.1 which have a TF Card.  This is a micro card reader.
The TF Card is connected via SPI (CS: GPIO13, MOSI; GPIO15, SCX: GPIO14 and MISO: GPIO02.

Can I use your custom component with this device. If so, what would be the custom_component configuration ?

n-serrette, 2025-02-18:
Hi,
I don't know if this board support mmc but you can try the following mapping:

```yaml
sd_mmc_card:
  id: sd_mmc_card
  mode_1bit: true
  clk_pin: GPIO14
  cmd_pin: GPIO15
  data0_pin: GPIO2
```

paradisgaston, 2025-02-18:
Thank you for your answer.
However, I don't know enough about external components. I only use it one
time with the following code in my program
"external_components:
  - source: github://swoboda1337/sx127x-esphome " Where I get the code
directly from the github at time of compile. Both examples that you give
are getting the component from a local file external_components:
  - source: components
I understand from ESPHome documentation that 'components' will be a
directory in my esphome directory.
Is there a way that I can get the necessary files at compile time as above?
If so, what will be the code.

If not, what files should I download and copy in my esphome/components
directory ?

n-serrette, 2025-02-18:
In the example, the external component source is the folder to avoid unnecessary clone of the repository during development.
You can use this component in the same way has the one already use, by setting the source to the repository url.

```yaml
external_components:
  source: github://n-serrette/esphome_sd_card
```

for more option you can look at the Esphome documentation on [external components](https://esphome.io/components/external_components)

deanfourie1, 2025-02-18:
Hi I am also trying to setup SD support on a LilyGo T-Call, my SD card module only have the following pins GND, VCC, MISO, MOSI, SCK, CS. I do not have any DATA pins.

Will this work and how do I wire it with your custom integration.

Thanks

n-serrette, 2025-02-20:
Hi,
So far I had no luck with a similar module, I don't know if my configuration is wrong or it's just not possible.
I don't know your specific card but you can try the [mapping](https://github.com/n-serrette/esphome_sd_card/blob/main/components/sd_mmc_card/README.md#config) provided by @elproko

deanfourie1, 2025-02-26:
> Hi, So far I had no luck with a similar module, I don't know if my configuration is wrong or it's just not possible. I don't know your specific card but you can try the [...]

Ok thanks, can you provide a link to a known working card reader.

Thanks

n-serrette, 2025-02-26:
I have only validated the module with an esp-cam at the moment, I plan to acquire and test this [reader](https://fr.aliexpress.com/item/1005004004351007.html) but I cannot give you a timeline.

deanfourie1, 2025-02-27:
> I have only validated the module with an esp-cam at the moment, I plan to acquire and test this [reader](https://fr.aliexpress.com/item/1005004004351007.html) but I cannot give you a timeline.

Right ok thanks. So would it be possible to read and write data from the LillyGO T-Call using a esp32cam module with the onboard SD Card?

n-serrette, 2025-02-28:
If you can communicate between the two esp it should be possible.

Will reading the esp-idf example [documentation](https://github.com/espressif/esp-idf/blob/master/examples/storage/sd_card/sdmmc/README.md#pin-assignments-for-esp32-s3), it is stated that a pull up on data3 is required for the sd card to work in mmc mode.

I will buy some resistor to try it out and will let you know the result