n-serrette/esphome_sd_card #12: esp32box3

youkorr, 2025-02-25:
Hi you did a great job, I would like to make it work on esp32s3box3 to mahere are the pins of the reader but i don't know how to make it workke the sd card work , cs = GPIO12     MISO = GPIO19   MOSI = GPIO14   SCLK = GPIO11

youkorr, 2025-02-25:
[https://github.com/espressif/esp-box/blob/master/hardware/SCH_ESP32-S3-BOX-3_V1.0/SCH_ESP32-S3-BOX-3-SENSOR-01_V1.1_20230922.pdf](url)

youkorr, 2025-02-26:
I set it up but on the logs I have this error it cannot detect the card `[01:54:21][C][sd_mmc_card:020]: SD MMC Component
[01:54:21][C][sd_mmc_card:021]:   Mode 1 bit: FALSE
[01:54:21][C][sd_mmc_card:022]:   CLK Pin: 11
[01:54:21][C][sd_mmc_card:023]:   CMD Pin: 14
[01:54:21][C][sd_mmc_card:024]:   DATA0 Pin: 9
[01:54:21][C][sd_mmc_card:026]:   DATA1 Pin: 13
[01:54:21][C][sd_mmc_card:027]:   DATA2 Pin: 42
[01:54:21][C][sd_mmc_card:028]:   DATA3 Pin: 12
[01:54:21][C][sd_mmc_card:031]:   Used space 'SD card used space'
[01:54:21][C][sd_mmc_card:031]:     State Class: 'measurement'
[01:54:21][C][sd_mmc_card:031]:     Unit of Measurement: 'B'
[01:54:21][C][sd_mmc_card:031]:     Accuracy Decimals: 2
[01:54:21][C][sd_mmc_card:031]:     Icon: 'mdi:memory'
[01:54:21][C][sd_mmc_card:032]:   Total space 'SD card total space'
[01:54:21][C][sd_mmc_card:032]:     State Class: 'measurement'
[01:54:21][C][sd_mmc_card:032]:     Unit of Measurement: 'B'
[01:54:21][C][sd_mmc_card:032]:     Accuracy Decimals: 2
[01:54:21][C][sd_mmc_card:032]:     Icon: 'mdi:memory'
[01:54:21][C][sd_mmc_card:040]:   SD Card Type 'SD card type'
[01:54:21][E][sd_mmc_card:044]: Setup failed : No card found
[01:54:21][E][component:082]:   Component sd_mmc_card is marked FAILED`

youkorr, 2025-02-27:
but the sd reader doesn't want to work

n-serrette, 2025-02-27:
Hi,
Sorry for the late reply, your configuration seems rights you can set `mode_1bit` to false to use all the data lanes.

Are you using the arduino framework or esp-idf?

Can you check that your sd card is formatted in FAT32?

youkorr, 2025-02-27:
thank you for answering me my sd card is formatted in fat32 with a test.mp3 and test.flac file at the root is I am in esp-idf

youkorr, 2025-02-27:
[E][sd_mmc_card:135]: Failed to open file for reading

youkorr, 2025-02-27:
`sd_mmc_card:
  id: sd_card
  mode_1bit: false
  clk_pin: GPIO11
  cmd_pin: GPIO14
  data0_pin: GPIO9
  data1_pin: GPIO13
  data2_pin: GPIO42  
  data3_pin: GPIO12`

youkorr, 2025-02-28:
[image]
[image]

youkorr, 2025-02-28:
[image]

youkorr, 2025-02-28:
[image]
[image]

youkorr, 2025-02-28:
I tried several combinations but the SD card does not want to mount.

youkorr, 2025-02-28:
[image]

n-serrette, 2025-02-28:
thanks for the details, could you try using the arduino framework,  the esp-idf is more recent and less tested.

youkorr, 2025-02-28:
the problem all my system works with esp-idf

youkorr, 2025-03-01:
hi and thanks for the answers but i think i found the problem i am missing the SDCARD_PWR_CTRL used for power control. it has a pin 43 to operate the sd reader but in yaml i can't put this
pwr_ctrl_pin: GPIO43 because your external_components doesn't allow it if you have a solution :)

[image]

n-serrette, 2025-03-03:
In the #7 you mentioned adding SDCARD_PWR_CTRL support, could you share the resources you have used so I can add it?

n-serrette, 2025-03-03:
you can add files and links in  issue,  if you have modified the code you can make a pull request, if you don't know how to use git you can put all your code in a zip file and I will integrate it.

youkorr, 2025-03-03:
`  // Enable SDCard power
  if (get_sdcard_power_ctrl_gpio() >= 0) {
    gpio_config_t gpio_cfg = {.mode = GPIO_MODE_OUTPUT, .pin_bit_mask = 1ULL << get_sdcard_power_ctrl_gpio()};
    gpio_config(&gpio_cfg);
    gpio_set_level(get_sdcard_power_ctrl_gpio(), 0);
  }`

youkorr, 2025-03-04:
Hi I will send you the whole code which works better, but I hope it doesn't bother you I had to associate with your code what you did

n-serrette, 2025-03-04:
Thanks, I gladly accept contributions, I made this project so other can benefit of it

youkorr, 2025-03-04:
I send the code mixed with yours for the ctrl power

youkorr, 2025-03-04:
[esp32_s3_box3_sd-main (1).zip](https://github.com/user-attachments/files/19075382/esp32_s3_box3_sd-main.1.zip)

youkorr, 2025-03-04:
do you have a discord?

n-serrette, 2025-03-04:
It's been a very long time since I last used discord, and I have very little free time at the moment.
Is their some things you which to discuss ?

You have removed the zip file, is their a reason?

youkorr, 2025-03-04:
yes to send you a pull request if I arrive

youkorr, 2025-03-04:
it doesn't work I'll send it to you directly

[esp32_s3_box3_sd-main (1).zip](https://github.com/user-attachments/files/19075541/esp32_s3_box3_sd-main.1.zip)

youkorr, 2025-03-04:
it's good for the file

youkorr, 2025-03-07:
what do you think about adding ctrl_pwr

n-serrette, 2025-03-07:
I have added the power_ctrl_pin option based on your code, I have implemented it in a more Esphome way (ie: using GPIOPin).

You can try it using the branch `power_control` #16

```yaml
external_components:
  - source: github://esphome/esphome@power_control
```

Once you have confirmed it is working i will merge the branch

youkorr, 2025-03-07:
ok i will test it

youkorr, 2025-03-07:
compoments??

youkorr, 2025-03-07:
unable to access 'https://esphome/esphome@power_control/': Could not resolve host: esphome.

n-serrette, 2025-03-07:
Sorry I have copy pasted the configuration from the doc without modifying it.
`source: github://n-serrette/esphome_sd_card@power_control`

youkorr, 2025-03-07:
ok

youkorr, 2025-03-07:
It would be interesting to have this yaml code in place  `storage:
  - platform: sd_card
    id: 
    files:
      - source:
        id: `

youkorr, 2025-03-07:
`[00:22:39][D][esp-idf:000]: I (2173) gpio: GPIO[43]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
[00:22:39]
[00:22:39][D][esp-idf:000]: I (2174) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
[00:22:39]
[00:22:39][D][esp-idf:000]: I (2175) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
[00:22:39]
[00:22:39][D][esp-idf:000]: I (2175) gpio: GPIO[9]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
[00:22:39]
[00:22:39][D][esp-idf:000]: I (2176) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
[00:22:39]
[00:22:39][D][esp-idf:000]: I (2176) gpio: GPIO[42]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
[00:22:39]
[00:22:39][D][esp-idf:000]: I (2177) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
[00:22:39]
[00:22:39][D][esp-idf:000]: I (2284) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 `

youkorr, 2025-03-07:
[image]

youkorr, 2025-03-07:
Perfect

youkorr, 2025-03-07:
Too bad I can't use the card reader to retrieve audio and image data to relieve esp32 with a lvgl display because the internal capacity is ridiculous

closed 2025-03-08 by n-serrette