n-serrette/esphome_sd_card #13: Webserver not working?

deanfourie1, 2025-02-28:
It appears to start up and everything, however I cannot access the webserver @ http://172.16.102.29/files, just times out.

Any ideas? Thanks

```
[22:01:16][I][app:100]: ESPHome version 2025.2.1 compiled on Feb 28 2025, 21:46:58
[22:01:16][C][wifi:600]: WiFi:
[22:01:16][C][wifi:428]:   Local MAC: 30:C6:F7:03:CC:14
[22:01:16][C][wifi:433]:   SSID: [redacted]
[22:01:16][C][wifi:436]:   IP Address: 172.16.102.29
[22:01:16][C][wifi:440]:   BSSID: [redacted]
[22:01:16][C][wifi:441]:   Hostname: 'esp32cam'
[22:01:16][C][wifi:443]:   Signal strength: -76 dB ▂▄▆█
[22:01:16][C][wifi:447]:   Channel: 11
[22:01:16][C][wifi:448]:   Subnet: 255.255.255.224
[22:01:16][C][wifi:449]:   Gateway: 172.16.102.1
[22:01:16][C][wifi:450]:   DNS1: 0.0.0.0
[22:01:16][C][wifi:451]:   DNS2: 0.0.0.0
[22:01:16][C][logger:177]: Logger:
[22:01:16][C][logger:178]:   Max Level: VERBOSE
[22:01:16][C][logger:179]:   Initial Level: VERBOSE
[22:01:16][C][logger:181]:   Log Baud Rate: 0
[22:01:16][C][logger:182]:   Hardware UART: UART0
[22:01:16][C][sd_file_server:016]: SD File Server:
[22:01:16][C][sd_file_server:017]:   Address: esp32cam.local:80
[22:01:16][C][sd_file_server:018]:   Url Prefix: file
[22:01:16][C][sd_file_server:019]:   Root Path: /
[22:01:17][C][sd_file_server:020]:   Deletation Enabled: TRUE
[22:01:17][C][sd_file_server:021]:   Download Enabled : TRUE
[22:01:17][C][sd_mmc_card:020]: SD MMC Component
[22:01:17][C][sd_mmc_card:021]:   Mode 1 bit: TRUE
[22:01:17][C][sd_mmc_card:022]:   CLK Pin: 14
[22:01:17][C][sd_mmc_card:023]:   CMD Pin: 15
[22:01:17][C][sd_mmc_card:024]:   DATA0 Pin: 2
[22:01:17][C][sd_mmc_card:031]:   Used space 'SD card used space'
[22:01:17][C][sd_mmc_card:031]:     State Class: 'measurement'
[22:01:17][C][sd_mmc_card:031]:     Unit of Measurement: 'Mb'
[22:01:17][C][sd_mmc_card:031]:     Accuracy Decimals: 2
[22:01:17][C][sd_mmc_card:031]:     Icon: 'mdi:memory'
[22:01:17][C][sd_mmc_card:032]:   Total space 'SD card total space'
[22:01:17][C][sd_mmc_card:032]:     State Class: 'measurement'
[22:01:17][C][sd_mmc_card:032]:     Unit of Measurement: 'Gb'
[22:01:17][C][sd_mmc_card:032]:     Accuracy Decimals: 2
[22:01:17][C][sd_mmc_card:032]:     Icon: 'mdi:memory'
[22:01:17][C][sd_mmc_card:033]:   Free space 'SD card free space'
[22:01:17][C][sd_mmc_card:033]:     State Class: 'measurement'
[22:01:17][C][sd_mmc_card:033]:     Unit of Measurement: 'Gb'
[22:01:17][C][sd_mmc_card:033]:     Accuracy Decimals: 2
[22:01:17][C][sd_mmc_card:033]:     Icon: 'mdi:memory'
[22:01:17][C][sd_mmc_card:036]:   File size 'text.txt size'
[22:01:17][C][sd_mmc_card:036]:     State Class: 'measurement'
[22:01:17][C][sd_mmc_card:036]:     Unit of Measurement: 'Kb'
[22:01:17][C][sd_mmc_card:036]:     Accuracy Decimals: 2
[22:01:17][C][sd_mmc_card:036]:     Icon: 'mdi:memory'
[22:01:17][C][sd_mmc_card:040]:   SD Card Type 'SD card type'
[22:01:17][C][captive_portal:089]: Captive Portal:
[22:01:17][C][mdns:116]: mDNS:
[22:01:17][C][mdns:117]:   Hostname: esp32cam
[22:01:17][V][mdns:118]:   Services:
```

esp32:
  board: esp32dev
  framework:
    type: arduino
external_components:
  - source:
      type: local
      path: custom_components
    components: [sd_mmc_card]

```
logger:
  level: VERBOSE #makes uart stream available in esphome logstream
  baud_rate: 0 #disable logging over uart
  on_message:
    level: DEBUG
    then:
      - sd_mmc_card.append_file:
          path: "/test.log" 
          data: !lambda |
            std::string str(message);
            str += "\n";
            return std::vector<uint8_t>(str.begin(), str.end());

sd_mmc_card:
  id: esp_camera_sd_card
  mode_1bit: true
  clk_pin: GPIO14
  cmd_pin: GPIO15
  data0_pin: GPIO2
  data1_pin: GPIO4
  data2_pin: GPIO12
  data3_pin: GPIO13

sensor:
  - platform: sd_mmc_card
    type: used_space
    unit_of_measurement: Mb
    name: "SD card used space"
    filters:
      - lambda: return sd_mmc_card::convertBytes(x, sd_mmc_card::MemoryUnits::MegaByte);

  - platform: sd_mmc_card
    type: total_space
    name: "SD card total space"
    unit_of_measurement: Gb
    filters:
      - lambda: return sd_mmc_card::convertBytes(x, sd_mmc_card::MemoryUnits::GigaByte);

  - platform: sd_mmc_card
    type: free_space
    name: "SD card free space"
    unit_of_measurement: Gb
    filters:
      - lambda: return sd_mmc_card::convertBytes(x, sd_mmc_card::MemoryUnits::GigaByte);

  - platform: sd_mmc_card
    type: file_size
    name: "text.txt size"
    unit_of_measurement: Kb
    path: "/test.txt"
    filters:
      - lambda: return sd_mmc_card::convertBytes(x, sd_mmc_card::MemoryUnits::KiloByte);

text_sensor:
  - platform: sd_mmc_card
    sd_card_type:
      name: "SD card type"

sd_file_server:
  id: file_server
  url_prefix: "file"
  root_path: "/"
  enable_deletion: true
  enable_download: true
  enable_upload: true
```

youkorr, 2025-03-02:
good evening a question that I try to understand how to put the ip address there  

`sd_file_server:
  id: file_server
  url_prefix: "file"
  root_path: "/"
  enable_deletion: true
  enable_download: true
  enable_upload: true`

I can't access it either and I know what to do, thanks

exemple 192.168.1.305

deanfourie1, 2025-03-03:
> good evening a question that I try to understand how to put the ip address there
> 
> `sd_file_server: id: file_server url_prefix: "file" root_path: "/" enable_deletion: true enable_download: true enable_upload: true`
> 
> I can't access it either and I know what to do, thanks
> 
> exemple 192.168.1.305

As I understand it just binds to the IP of the esp

youkorr, 2025-03-03:
ok thanks for your answer I had already tried to connect to the esp ip address but it doesn't work because I don't have the same device mine is connected with home assistant

n-serrette, 2025-03-03:
As @deanfourie1 stated the file server run on the esp to access the web page you have to use the esp ip.

The component is not intended to be integrated in HA, so their is no straightforward way to do it.

From here you have to way:
* using HTTP request to query the esp but the API is incomplete 
* using a webpage card in your dashboard pointing on http://<esp-ip>/<url-prefix>

youkorr, 2025-03-03:
Thank you for your reply, have you tried it yet?

n-serrette, 2025-03-03:
I have tried the webpage card option and it work, just a side note deleting and uploading files does not work on esp-idf

n-serrette, 2025-03-04:
could give me the esp setup logs to see more details.

Also their is some issues in with the recommended arduino framework version, could you switch to the latest.

```yaml
esp32:
  board: esp32dev
  framework:
    type: arduino
    version: latest
```

deanfourie1, 2025-03-04:
> esp32:
>   board: esp32dev
>   framework:
>     type: arduino
>     version: latest

Tried this, no luck.

Sure, here are the setup logs.

[09:54:28][I][app:100]: ESPHome version 2025.2.2 compiled on Mar  5 2025, 09:52:31
[09:54:28][C][wifi:600]: WiFi:
[09:54:28][C][wifi:428]:   Local MAC: [redacted]
[09:54:28][C][wifi:433]:   SSID: [redacted]
[09:54:28][C][wifi:436]:   IP Address: [redacted]
[09:54:28][C][wifi:440]:   BSSID: [redacted]
[09:54:28][C][wifi:441]:   Hostname: 'esp32cam'
[09:54:29][C][wifi:443]:   Signal strength: -64 dB ▂▄▆█
[09:54:29][C][wifi:447]:   Channel: 11
[09:54:29][C][wifi:448]:   Subnet: [redacted]
[09:54:29][C][wifi:449]:   Gateway: [redacted]
[09:54:29][C][wifi:450]:   DNS1: 0.0.0.0
[09:54:29][C][wifi:451]:   DNS2: 0.0.0.0
[09:54:29][C][logger:177]: Logger:
[09:54:29][C][logger:178]:   Max Level: VERBOSE
[09:54:29][C][logger:179]:   Initial Level: VERBOSE
[09:54:29][C][logger:181]:   Log Baud Rate: 0
[09:54:29][C][logger:182]:   Hardware UART: UART0
[09:54:29][C][sd_file_server:016]: SD File Server:
[09:54:29][C][sd_file_server:017]:   Address: esp32cam.local:80
[09:54:29][C][sd_file_server:018]:   Url Prefix: files
[09:54:29][C][sd_file_server:019]:   Root Path: /
[09:54:29][C][sd_file_server:020]:   Deletation Enabled: TRUE
[09:54:29][C][sd_file_server:021]:   Download Enabled : TRUE
[09:54:29][C][sd_mmc_card:020]: SD MMC Component
[09:54:29][C][sd_mmc_card:021]:   Mode 1 bit: TRUE
[09:54:29][C][sd_mmc_card:022]:   CLK Pin: 14
[09:54:29][C][sd_mmc_card:023]:   CMD Pin: 15
[09:54:29][C][sd_mmc_card:024]:   DATA0 Pin: 2
[09:54:29][C][sd_mmc_card:031]:   Used space 'SD card used space'
[09:54:29][C][sd_mmc_card:031]:     State Class: 'measurement'
[09:54:29][C][sd_mmc_card:031]:     Unit of Measurement: 'Mb'
[09:54:29][C][sd_mmc_card:031]:     Accuracy Decimals: 2
[09:54:29][C][sd_mmc_card:031]:     Icon: 'mdi:memory'
[09:54:29][C][sd_mmc_card:032]:   Total space 'SD card total space'
[09:54:29][C][sd_mmc_card:032]:     State Class: 'measurement'
[09:54:29][C][sd_mmc_card:032]:     Unit of Measurement: 'Gb'
[09:54:29][C][sd_mmc_card:032]:     Accuracy Decimals: 2
[09:54:29][C][sd_mmc_card:032]:     Icon: 'mdi:memory'
[09:54:29][C][sd_mmc_card:033]:   Free space 'SD card free space'
[09:54:29][C][sd_mmc_card:033]:     State Class: 'measurement'
[09:54:29][C][sd_mmc_card:033]:     Unit of Measurement: 'Gb'
[09:54:29][C][sd_mmc_card:033]:     Accuracy Decimals: 2
[09:54:29][C][sd_mmc_card:033]:     Icon: 'mdi:memory'
[09:54:29][C][sd_mmc_card:036]:   File size 'text.txt size'
[09:54:30][C][sd_mmc_card:036]:     State Class: 'measurement'
[09:54:30][C][sd_mmc_card:036]:     Unit of Measurement: 'Kb'
[09:54:30][C][sd_mmc_card:036]:     Accuracy Decimals: 2
[09:54:30][C][sd_mmc_card:036]:     Icon: 'mdi:memory'
[09:54:30][C][sd_mmc_card:040]:   SD Card Type 'SD card type'
[09:54:30][C][captive_portal:089]: Captive Portal:
[09:54:30][C][mdns:116]: mDNS:
[09:54:30][C][mdns:117]:   Hostname: esp32cam
[09:54:30][V][mdns:118]:   Services:
[09:54:30][V][mdns:120]:   - _esphomelib, _tcp, 6053
[09:54:30][V][mdns:122]:     TXT: friendly_name = esp32cam
[09:54:30][V][mdns:122]:     TXT: version = 2025.2.2
[09:54:30][V][mdns:122]:     TXT: mac = 30c6f703cc14
[09:54:30][V][mdns:122]:     TXT: platform = ESP32
[09:54:30][V][mdns:122]:     TXT: board = esp32dev
[09:54:30][V][mdns:122]:     TXT: network = wifi
[09:54:30][V][mdns:122]:     TXT: api_encryption = Noise_NNpsk0_25519_ChaChaPoly_SHA256
[09:54:30][C][esphome.ota:073]: Over-The-Air updates:
[09:54:30][C][esphome.ota:074]:   Address: esp32cam.local:3232
[09:54:30][C][esphome.ota:075]:   Version: 2
[09:54:30][C][esphome.ota:078]:   Password configured
[09:54:30][C][safe_mode:018]: Safe Mode:
[09:54:30][C][safe_mode:020]:   Boot considered successful after 60 seconds
[09:54:30][C][safe_mode:021]:   Invoke after 10 boot attempts
[09:54:30][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[09:54:30][C][api:140]: API Server:
[09:54:30][C][api:141]:   Address: esp32cam.local:6053
[09:54:30][C][api:143]:   Using noise encryption: YES

n-serrette, 2025-03-07:
Can you try adding the webserver component in your config?

```yaml
web_server:

```

deanfourie1, 2025-03-10:
I added the webserver: component and still does not work,

However, I did stumble across something strange. With no SD card inserted, I was able to see a webpage showing an error.

Something along the lines of "Unable to access files", again, with the SD inserted, I get no webserver at all.

youkorr, 2025-03-10:
Hi, do a test with a fixed address that you don't use

youkorr, 2025-03-10:
`external_components:
  - source:
      type: local
      path: custom_components
    components: [sd_mmc_card, sd_file_server]`

youkorr, 2025-03-10:
`wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip: 
    static_ip: xxxxxxxxxxxxx
    gateway: xxxxxxxxxxxx
    subnet: 255.255.255.0`

deanfourie1, 2025-03-11:
I already am using a static IP.

Yes, the sensors do report the value of the SD card such as, total space, space left etc.

youkorr, 2025-03-11:
as he had serette for sd_file_server to work you must put web server, I just did the test that you did but by adding web server it works👍

deanfourie1, 2025-03-11:
Hmm I tried that but its not working,

I am going to try format the SD again, 

FAT32 right?

youkorr, 2025-03-11:
yes in fat32, are you sure that you would not be mistaken on the pins

youkorr, 2025-03-11:
sd_mmc_card:
  id: esp_camera_sd_card
  mode_1bit: false
  clk_pin: GPIO14
  cmd_pin: GPIO15
  data0_pin: GPIO2
  data1_pin: GPIO4
  data2_pin: GPIO12
  data3_pin: GPIO13

for 4 bit you have to put mode_1bit: false

deanfourie1, 2025-03-12:
Ok, 

I formatted in fat32 and now its working, I think the fix was

`webserver:`

Thanks all for your help with this.

youkorr, 2025-03-12:
Hi, I suspect a hardware problem because I don't see it.

youkorr, 2025-03-12:
can you tell me what the power pin is

deanfourie1, 2025-03-13:
This is my setup

```
sd_mmc_card:
  id: esp_camera_sd_card
  mode_1bit: true
  clk_pin: GPIO14
  cmd_pin: GPIO15
  data0_pin: GPIO2
  data1_pin: GPIO4
  data2_pin: GPIO12
  data3_pin: GPIO13
```

n-serrette, 2025-03-13:
In 1 bit mode the pin `data1_pin`, `data2_pin`, `data3_pin` are not used.

If you don't have more issue with the web_sever can I close the issue?

deanfourie1, 2025-03-16:
yes I think you can close now thanks

closed 2025-03-17 by n-serrette