n-serrette/esphome_sd_card #15: ESP32-S3 support?

krystof-k, 2025-03-06:
Hey there, I can't make it work for ESP32-S3 (using AI-S3 board). I get `Failed to mount card` no matter the pins I use.

This is my current config:

```yaml
external_components:
  source: github://n-serrette/esphome_sd_card

sd_mmc_card:
  id: card
  mode_1bit: true
  clk_pin: GPIO36
  cmd_pin: GPIO35
  data0_pin: GPIO37
  # data1_pin: GPIO4
  # data2_pin: GPIO12
  # data3_pin: GPIO13

sd_file_server:
  id: file_server
  url_prefix: file
  root_path: "/"
  enable_deletion: true
  enable_download: true
  enable_upload: true
```

I have [this SD card reader](https://www.laskakit.cz/en/sd-card-modul-spi/), CS is connected to GPIO1. When using ESP32 (WT32-ETH01) it worked fine.

But the result is this:

```
[11:16:35][C][sd_mmc_card:020]: SD MMC Component
[11:16:35][C][sd_mmc_card:021]:   Mode 1 bit: TRUE
[11:16:35][C][sd_mmc_card:022]:   CLK Pin: 36
[11:16:35][C][sd_mmc_card:023]:   CMD Pin: 35
[11:16:35][C][sd_mmc_card:024]:   DATA0 Pin: 37
[11:16:35][E][sd_mmc_card:044]: Setup failed : Failed to mount card
[11:16:35][E][component:082]:   Component sd_mmc_card is marked FAILED
```

Any idea how to do it?

youkorr, 2025-03-06:
you are missing data3

krystof-k, 2025-03-06:
Could you be more specific, please? I'm using SPI mode, if I add `data3_pin`, it doesn't havy any effect:

https://github.com/n-serrette/esphome_sd_card/blob/f4a5bb8eacdb1bda0ad777694c23f28048479031/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp#L17

youkorr, 2025-03-06:
[image]

youkorr, 2025-03-06:
does it work??

krystof-k, 2025-03-06:
I will retry that again later, but based on the code adding the `data3_pin` doesn't have any effeect in 1-bit mode. However I just found out that I probably need to [add a 10k pullup](https://github.com/espressif/arduino-esp32/tree/master/libraries/SD_MMC#pin-assignments-for-esp32-s3) to the CS pin.

krystof-k, 2025-03-06:
Unfortunately no, it's not working. I tried this:

```yaml
sd_mmc_card:
  id: card
  mode_1bit: true
  clk_pin: GPIO36
  cmd_pin: GPIO35
  data0_pin: GPIO37
  data3_pin: GPIO38
```

And adding pull-ups to all CS, MOSI,SCK and MISO, but it still shows the same `Failed to mount card` error.

youkorr, 2025-03-06:
Is the card reader properly powered?

krystof-k, 2025-03-11:
I can't make it work. It works only with ESP32-CAM. Any board I use and connect any SD card reader, I get either `Failed to set pins` or `Failed to mount card`. I really have no idea what am I doing wrong or where is the problem.

youkorr, 2025-03-11:
you have a pin error it's not the right pin

krystof-k, 2025-03-11:
I have tried everything. I have brand new [microSD reader module](https://www.laskakit.cz/en/laskakit-microsd-card-modul/), connected like this:

```
black GND → GND
red VCC → 3V3
blue SDO (DATA0) → GPIO2
yellow SCK (CLK) → GPIO14
white SDI (CMD) → GPIO15
```

CS (DATA3) pin has 3V on the module, as I measured and verified.

This is the config:

```yaml
external_components:
  source: github://n-serrette/esphome_sd_card

sd_mmc_card:
  id: storage
  mode_1bit: true
  clk_pin: GPIO14
  cmd_pin: GPIO15
  data0_pin: GPIO2
```

I tried both ESP32 and ESP32-S3 boards and I always get `Failed to mount card`.

Exactly the same config on ESP32-CAM works:

```
[18:07:38][C][sd_file_server:016]: SD File Server:
[18:07:38][C][sd_file_server:017]:   Address: cam.local:80
[18:07:38][C][sd_file_server:018]:   Url Prefix: file
[18:07:38][C][sd_file_server:019]:   Root Path: /
[18:07:38][C][sd_file_server:020]:   Deletation Enabled: TRUE
[18:07:38][C][sd_file_server:021]:   Download Enabled : TRUE
[18:07:38][C][sd_file_server:022]:   Upload Enabled : TRUE
[18:07:38][C][sd_mmc_card:020]: SD MMC Component
[18:07:38][C][sd_mmc_card:021]:   Mode 1 bit: TRUE
[18:07:38][C][sd_mmc_card:022]:   CLK Pin: 14
[18:07:38][C][sd_mmc_card:023]:   CMD Pin: 15
[18:07:38][C][sd_mmc_card:024]:   DATA0 Pin: 2
```

n-serrette, 2025-03-11:
the `Failed to mount card` error indicate that the pin used are the correct ones for the esp.

In your first post you were saying that you add some success with your WT32-ETH01 but later you were unable to reproduce the setup, could you detail your original setup?

I also have some issues with board other than the esp cam, I have ordered some component for testing, I will let you know the results

krystof-k, 2025-03-12:
OK, so I tried the WT32-ETH01 board again and it works. With the microSD card module, SD card module, I even tried to solder pins directly to microSD card adapter (+ pull-ups). All works. With all other boards (except ESP32-CAM with the onboard slot), nothing works.

So I assume there has to be something wrong within the internal architecture of certain boards, maybe related to power, but I even tried powering the SD module directly with 5 V and it didn't help.

krystof-k, 2025-03-12:
What do you mean by power supply GPIO?

youkorr, 2025-03-12:
Some SD card readers can be powered by a GPIO pin.

youkorr, 2025-03-12:
on my esp32s3box3 is powered by a crtl_pwr so esp32s3 must have

youkorr, 2025-03-12:
`/**
 * @brief SDCARD Function Definition
 */
#define FUNC_SDCARD_EN              (1)
#define SDCARD_OPEN_FILE_NUM_MAX    (5)
#define SDCARD_INTR_GPIO            (-1)
#define SDCARD_PWR_CTRL             GPIO_NUM_43
#define ESP_SD_PIN_CLK              GPIO_NUM_11
#define ESP_SD_PIN_CMD              GPIO_NUM_14
#define ESP_SD_PIN_D0               GPIO_NUM_9
#define ESP_SD_PIN_D1               GPIO_NUM_13
#define ESP_SD_PIN_D2               GPIO_NUM_42
#define ESP_SD_PIN_D3               GPIO_NUM_12
#define ESP_SD_PIN_D4               (-1)
#define ESP_SD_PIN_D5               (-1)
#define ESP_SD_PIN_D6               (-1)
#define ESP_SD_PIN_D7               (-1)
#define ESP_SD_PIN_CD               (-1)
#define ESP_SD_PIN_WP               (-1)`

youkorr, 2025-03-12:
in sd_mmc because you have an option that you can put power_crtl_pin:
`sd_mmc_card:
  id: sd_card
  clk_pin: GPIO11
  cmd_pin: GPIO14
  data0_pin: GPIO9
  data1_pin: GPIO13
  data2_pin: GPIO42
  data3_pin: GPIO12
  mode_1bit: false
  power_ctrl_pin: GPIO43`