n-serrette/esphome_sd_card #19: ESP32-S2 Mini: Error compiling

TCB13, 2025-03-23:
Hello,

I'm not sure if this related to my board or something on my config but I'm unable to compile it.

Configuration:

```
esphome:
  name: esp-sala
  friendly_name: esp-sala

external_components:
  - source: github://n-serrette/esphome_sd_card@main
    refresh: 0s

esp32:
  board: lolin_s2_mini
  variant: ESP32S2
  framework:
    type: arduino

logger:
  level: DEBUG

// (...) api...ota..wifi..webserver config here

sd_mmc_card:
  id: sd_mmc_card_sala
  mode_1bit: true
  clk_pin: GPIO38
  cmd_pin: GPIO39
  data0_pin: GPIO37
  ##data1_pin: GPIO4
  ##data2_pin: GPIO12
  #data3_pin: GPIO40

(...)
```

Compilation output:

```
INFO ESPHome 2025.3.0
INFO Reading configuration webui-config\esp-sala.yaml...
INFO Updating https://github.com/n-serrette/esphome_sd_card.git@main
INFO Generating C++ source...
INFO Compiling app...
Processing esp-sala (board: lolin_s2_mini; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.3.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.6
|-- Wire @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- SD_MMC @ 2.0.0
|-- SPI @ 2.0.0
|-- BME68x Sensor library @ 1.1.40407
|-- bsec2 @ 1.8.2610+sha.24eccb8
Compiling .pioenvs\esp-sala\src\esphome\components\sd_mmc_card\sd_mmc_card_esp32_arduino.cpp.o
Compiling .pioenvs\esp-sala\src\main.cpp.o
Archiving .pioenvs\esp-sala\lib592\libWiFi.a
Indexing .pioenvs\esp-sala\lib592\libWiFi.a
Archiving .pioenvs\esp-sala\libcbd\libFS.a
Indexing .pioenvs\esp-sala\libcbd\libFS.a
Archiving .pioenvs\esp-sala\lib867\libUpdate.a
Indexing .pioenvs\esp-sala\lib867\libUpdate.a
Compiling .pioenvs\esp-sala\libf1f\ESPAsyncWebServer-esphome\AsyncWebSocket.cpp.o
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp: In member function 'virtual void esphome::sd_mmc_card::SdMmc::setup()':
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp:20:42: error: 'SD_MMC' was not declared in this scope
   bool setPinResult = this->mode_1bit_ ? SD_MMC.setPins(this->clk_pin_, this->cmd_pin_, this->data0_pin_)
                                          ^~~~~~
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp:44:19: error: 'CARD_NONE' was not declared in this scope
   if (cardType == CARD_NONE) {
                   ^~~~~~~~~
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp:44:19: note: suggested alternative: 'PARSE_NONE'
   if (cardType == CARD_NONE) {
                   ^~~~~~~~~
                   PARSE_NONE
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp: In member function 'void esphome::sd_mmc_card::SdMmc::write_file(const char*, const uint8_t*, size_t, const char*)':
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp:54:15: error: 'SD_MMC' was not declared in this scope
   File file = SD_MMC.open(path, mode);
               ^~~~~~
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp: In member function 'bool esphome::sd_mmc_card::SdMmc::create_directory(const char*)':
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp:67:8: error: 'SD_MMC' was not declared in this scope
   if (!SD_MMC.mkdir(path)) {
        ^~~~~~
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp: In member function 'bool esphome::sd_mmc_card::SdMmc::remove_directory(const char*)':
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp:77:8: error: 'SD_MMC' was not declared in this scope
   if (!SD_MMC.rmdir(path)) {
        ^~~~~~
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp: In member function 'bool esphome::sd_mmc_card::SdMmc::delete_file(const char*)':
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp:87:8: error: 'SD_MMC' was not declared in this scope
   if (!SD_MMC.remove(path)) {
        ^~~~~~
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp: In member function 'std::vector<unsigned char> esphome::sd_mmc_card::SdMmc::read_file(const char*)':
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp:97:15: error: 'SD_MMC' was not declared in this scope
   File file = SD_MMC.open(path);
               ^~~~~~
Compiling .pioenvs\esp-sala\libf1f\ESPAsyncWebServer-esphome\WebAuthentication.cpp.o
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp: In member function 'std::vector<esphome::sd_mmc_card::FileInfo>& esphome::sd_mmc_card::SdMmc::list_directory_file_info_rec(const char*, uint8_t, std::vector<esphome::sd_mmc_card::FileInfo>&)':
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp:115:15: error: 'SD_MMC' was not declared in this scope
   File root = SD_MMC.open(path);
               ^~~~~~
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp: In member function 'bool esphome::sd_mmc_card::SdMmc::is_directory(const char*)':
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp:139:15: error: 'SD_MMC' was not declared in this scope
   File root = SD_MMC.open(path);
               ^~~~~~
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp: In member function 'size_t esphome::sd_mmc_card::SdMmc::file_size(const char*)':
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp:148:15: error: 'SD_MMC' was not declared in this scope
   File file = SD_MMC.open(path);
               ^~~~~~
Compiling .pioenvs\esp-sala\libf1f\ESPAsyncWebServer-esphome\WebHandlers.cpp.o
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp: In member function 'std::__cxx11::string esphome::sd_mmc_card::SdMmc::sd_card_type_to_string(int) const':
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp:154:10: error: 'CARD_NONE' was not declared in this scope
     case CARD_NONE:
          ^~~~~~~~~
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp:154:10: note: suggested alternative: 'PARSE_NONE'
     case CARD_NONE:
          ^~~~~~~~~
          PARSE_NONE
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp:156:10: error: 'CARD_MMC' was not declared in this scope
     case CARD_MMC:
          ^~~~~~~~
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp:156:10: note: suggested alternative: 'RAND_MAX'
     case CARD_MMC:
          ^~~~~~~~
          RAND_MAX
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp:158:10: error: 'CARD_SD' was not declared in this scope
     case CARD_SD:
          ^~~~~~~
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp:160:10: error: 'CARD_SDHC' was not declared in this scope
     case CARD_SDHC:
          ^~~~~~~~~
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp:162:10: error: 'CARD_UNKNOWN' was not declared in this scope
     case CARD_UNKNOWN:
          ^~~~~~~~~~~~
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp:162:10: note: suggested alternative: 'FM_UNKNOWN'
     case CARD_UNKNOWN:
          ^~~~~~~~~~~~
          FM_UNKNOWN
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp: In member function 'void esphome::sd_mmc_card::SdMmc::update_sensors()':
src/esphome/components/sd_mmc_card/sd_mmc_card_esp32_arduino.cpp:170:25: error: 'SD_MMC' was not declared in this scope
   uint64_t used_bytes = SD_MMC.usedBytes();
                         ^~~~~~
Compiling .pioenvs\esp-sala\libf1f\ESPAsyncWebServer-esphome\WebRequest.cpp.o
Compiling .pioenvs\esp-sala\libf1f\ESPAsyncWebServer-esphome\WebResponses.cpp.o
Compiling .pioenvs\esp-sala\libf1f\ESPAsyncWebServer-esphome\WebServer.cpp.o
Compiling .pioenvs\esp-sala\libabc\ESPmDNS\ESPmDNS.cpp.o
*** [.pioenvs\esp-sala\src\esphome\components\sd_mmc_card\sd_mmc_card_esp32_arduino.cpp.o] Error 1
========================== [FAILED] Took 9.26 seconds ==========================
```

Thank you.

n-serrette, 2025-03-24:
Hi,

I was able to reproduce your issue, their is also an issue when compiling with esp-idf, I will look into it.
I don't see anything wrong with your config so it may be due to the board.

TCB13, 2025-03-24:
> Hi,
> 
> I was able to reproduce your issue, their is also an issue when compiling with esp-idf, I will look into it. I don't see anything wrong with your config so it may be due to the board.

Much appreciated.

TCB13, 2025-03-24:
Thanks checking, but isn't `mode_1bit` actually an SPI implementation already? I'm confused.

n-serrette, 2025-03-24:
It's not the same thing, in 1 bit mode the sdmmc driver is still used.

I'm not totaly clear on the inner working of the sdmmc and spi driver

TCB13, 2025-03-24:
Okay, all right. I've noticed this also applies to the ESP32-C3 as well.

TCB13, 2026-02-18:
Related: https://github.com/n-serrette/esphome_sd_card/pull/25