n-serrette/esphome_sd_card #20: GPIO expanders do not appear to work for CS pin

cryptk, 2025-04-07:
It does not appear that this component allows for GPIO expanders to be used for the various pins.  While this is likely a poor design choice if the hardware uses one for the MOSI/MISO pins, there is a pretty valid design choice for putting the CS pin on a GPIO expander.  I have one such dev board.

With the following configuration:

```
sd_mmc_card:
  id: sd_card
  mode_1bit: true
  clk_pin: GPIO2  # CLK/SCK in SPI mode
  cmd_pin: GPIO1  # MOSI in SPI mode
  data0_pin: GPIO42  # MISO in SPI mode
  data3_pin:   # SS/CS in SPI mode
    pca9554: gpio_expander
    number: 3
```

I receive the following error when attempting to compile:
```
INFO ESPHome 2025.3.3
INFO Reading configuration config/waveshare-2.8-round.yaml...
Failed config

sd_mmc_card: [source config/2.8_round_alarm_clock_packages/sdcard.yaml:2]
  id: sd_card
  mode_1bit: True
  clk_pin: GPIO2
  cmd_pin: GPIO1
  data0_pin: GPIO42
  data3_pin: 
    
    [pca9554] is an invalid option for [data3_pin]. Please check the indentation.
    pca9554: gpio_expander
    number: 3
```

cryptk, 2025-04-09:
I have it working on an SPI based interface on a module made by waveshare.  I changed the schema for data3_pin to `cv.Optional(CONF_DATA3_PIN): pins.gpio_pin_schema({CONF_OUTPUT: True}),` to allow me to supply the config like this:

```
  data3_pin: # SS/CS in SPI mode
    pca9554: gpio_expander
    number: 3
```

I also had to make one other modification to work around my other issue, but so far it seems to be working.

gebbe75, 2026-01-06:
I want to use a esp32-s3-touch-lcd-7b dislay with waveshare_io_ch32v003 expander but it does not sees to work with these expander.

configuration: 
...
sd_mmc_card:
  id: sd_mmc_card_intern
  mode_1bit: true
  clk_pin: GPIO12
  cmd_pin: GPIO11
  data0_pin: GPIO13
  data3_pin: # SS/CS on expander EXIO4 (PD4)
    waveshare_io_ch32v003: waveshare_io_hub
    number: 4
...

I recived the error: 

 [waveshare_io_ch32v003] is an invalid option for [data3_pin]. Please check the indentation.